osrf/gzweb #183: npm build/deploy error

anthonypan08, 2020-06-18:
I am following [this](http://gazebosim.org/tutorials?tut=gzweb_install&cat=gzweb) installation guide but ran into some issues at the npm step on build. Below is the error I am getting \
[image].

I am running gazebo 9 with node version v8.17.0 and npm version 6.13.4. I tried both build on gzweb_1.4.0 and master

reinzor, 2020-06-18:
Your `configure` step already seems to fail due to an already existing file. Please provide a proper log of the full clean `./deploy.sh` script.

anthonypan08, 2020-06-18:
I think I figured it out, apparently I didn't install npm-gyp.

closed 2020-06-18 by anthonypan08