osrf/gzweb #184: Compilation with gazebo11

reinzor, 2020-06-18:
The `1.4.0` branch is not working with Gazebo 11 on Ubuntu Focal.

reinzor, 2020-06-18:
https://github.com/eurogroep/gzweb/commit/696148dc49b8a0ebd54d749c2e1a0d28189de120 makes gzweb compile with C++17 and some boost fixes. I don't know how to make this backwards compatible properly due to the lack of knowledge of `node` `gyp`.

chapulina, 2020-06-18:
Thanks for figuring this out. We could create a new branch that's compatible with Gazebo 11.

We should also update the tutorial: http://gazebosim.org/tutorials?tut=gzweb_install&cat=gzweb

Jaeyoung-Lim, 2020-10-06:
@reinzor @chapulina Are there any updates regarding this topic?

reinzor, 2020-10-06:
Nope, we have been using [this fix](https://github.com/eurogroep/gzweb/commit/696148dc49b8a0ebd54d749c2e1a0d28189de120) up until now.

chapulina, 2020-10-06:
@reinzor , do you know a way to make your changes compatible with Bionic / Gazebo 9? If so, we could put it in the master branch to support both 9 and 11.

reinzor, 2020-10-06:
Sorry, I am a little out of my comfort zone here with `node` `gyp`.

iche033, 2020-10-06:
I applied @reinzor's fix in pull request #194. I didn't have to apply the `sed` boost cmd in `deploy.sh` and it still built so I left that out.

mis-eu, 2020-10-07:
hi @iche033 
do you mind linking here the precise repo/branch source code you cloned to get [reiznor's gazebo11 fix](https://github.com/eurogroep/gzweb/commit/696148dc49b8a0ebd54d749c2e1a0d28189de120) to work?

you had success compiling gzweb for Gazebo11 on Bionic

how about compiling gzweb for Gazebo11 on Foxy/Focal?

mis-eu, 2020-10-07:
> Nope, we have been using [this fix](https://github.com/eurogroep/gzweb/commit/696148dc49b8a0ebd54d749c2e1a0d28189de120) up until now.

hi @reinzor 
great patch, although does not seem to compile GzWeb / Gazebo11 on Foxy/Focal
which system releases are you on?

reinzor, 2020-10-07:
> which system releases are you on?

Gazebo 11 on Ubuntu Focal

mis-eu, 2020-10-12:
@reinzor and which version node and npm?  
Failed with:
node v.12.18.4
npm 6.14.6

**Also failed with @iche033 suggested:
node v6.17.1
npm 3.10.10**

Failed to build on Ubuntu Focal/Foxy/Gazebo11 the https://github.com/osrf/gzweb checked out 1.4
with manually added both patches: https://github.com/eurogroep/gzweb/commit/696148dc49b8a0ebd54d749c2e1a0d28189de120

I even installed Boost from source. Would inking the Boost lib to the gzweb build be required?

<h3>gzweb Focal/Foxy build error Ouput below</h3>
<details>
user@machine:~/gzweb$ npm run deploy --- -m

> gzweb@1.3.0 deploy /home/user/gzweb
> ./deploy.sh "-m"

Build a local model database.
Download from gazebo_models repository.
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: fsevents@^1.0.0 (node_modules/chokidar/node_modules/fsevents):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for fsevents@1.2.13: wanted {"os":"darwin","arch":"any"} (current: {"os":"linux","arch":"x64"})
npm WARN ajv-keywords@2.1.1 requires a peer of ajv@^5.0.0 but none was installed.
npm WARN gzweb@1.3.0 No license field.
Running "concat:build_src" (concat) task

Running "concat:build_gz3d" (concat) task

Running "concat:build_gui" (concat) task

Running "jshint:files" (jshint) task
>> 2 files lint free.

Running "uglify:build_src" (uglify) task
>> 1 file created 1.48 MB → 693.93 kB

Running "uglify:build_gz3d" (uglify) task
>> 1 file created 1.48 MB → 693.93 kB

Running "uglify:build_gui" (uglify) task
>> 1 file created 2.13 MB → 1.14 MB

Done, without errors.
-- The C compiler identification is GNU 9.3.0
-- The CXX compiler identification is GNU 9.3.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- gzweb version 2.0.0
CMake Warning (dev) at CMakeLists.txt:23 (set):
  Policy CMP0053 is not set: Simplify variable reference and escape sequence
  evaluation.  Run "cmake --help-policy CMP0053" for policy details.  Use the
  cmake_policy command to set the policy and suppress this warning.

  For input:

    '@PROJECT_NAME@-@GZWEB_VERSION_FULL@'

[... 256 lines cut ...]
There are node-gyp build errors, exiting.

npm ERR! Linux 4.19.104-microsoft-standard
npm ERR! argv "/home/user/.nvm/versions/node/v6.17.1/bin/node" "/home/user/.nvm/versions/node/v6.17.1/bin/npm" "run" "deploy" "---" "-m"
npm ERR! node v6.17.1
npm ERR! npm  v3.10.10
npm ERR! code ELIFECYCLE
npm ERR! gzweb@1.3.0 deploy: `./deploy.sh "-m"`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the gzweb@1.3.0 deploy script './deploy.sh "-m"'.
npm ERR! Make sure you have the latest version of node.js and npm installed.
npm ERR! If you do, this is most likely a problem with the gzweb package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     ./deploy.sh "-m"
npm ERR! You can get information on how to open an issue for this project with:
npm ERR!     npm bugs gzweb
npm ERR! Or if that isn't available, you can get their info via:
npm ERR!     npm owner ls gzweb
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     /home/user/gzweb/npm-debug.log
</deploy>

mis-eu, 2020-10-12:
and below is the **Makefile** log

<details>
# CMAKE generated file: DO NOT EDIT!
# Generated by "Unix Makefiles" Generator, CMake Version 3.16

# Default target executed when no arguments are given to make.
default_target: all

.PHONY : default_target

# Allow only one "make -f Makefile2" at a time, but pass parallelism.
.NOTPARALLEL:

#=============================================================================
# Special targets provided by cmake.

# Disable implicit rules so canonical targets will work.
.SUFFIXES:

# Remove some rules from gmake that .SUFFIXES does not remove.
SUFFIXES =

.SUFFIXES: .hpux_make_needs_suffix_list

# Suppress display of executed commands.
$(VERBOSE).SILENT:

# A target that is always out of date.
cmake_force:

.PHONY : cmake_force

#=============================================================================
# Set environment variables for the build.

# The shell in which to execute make rules.
SHELL = /bin/sh

# The CMake executable.
CMAKE_COMMAND = /usr/bin/cmake

# The command to remove a file.
RM = /usr/bin/cmake -E remove -f

# Escaping for special characters.
EQUALS = =

# The top-level source directory on which CMake was run.
CMAKE_SOURCE_DIR = /home/esofi/gzweb

# The top-level build directory on which CMake was run.
CMAKE_BINARY_DIR = /home/esofi/gzweb/build

#=============================================================================
# Targets provided globally by CMake.

# Special rule for the target edit_cache
edit_cache:
	@$(CMAKE_COMMAND) -E cmake_echo_color --switch=$(COLOR) --cyan "No interactive CMake dialog available..."
	/usr/bin/cmake -E echo No\ interactive\ CMake\ dialog\ available.
.PHONY : edit_cache

# Special rule for the target edit_cache
edit_cache/fast: edit_cache

.PHONY : edit_cache/fast

# Special rule for the target test
test:
	@$(CMAKE_COMMAND) -E cmake_echo_color --switch=$(COLOR) --cyan "Running tests..."
	/usr/bin/ctest --force-new-ctest-process $(ARGS)
.PHONY : test

# Special rule for the target test
test/fast: test

.PHONY : test/fast

# Special rule for the target package_source
package_source:
	@$(CMAKE_COMMAND) -E cmake_echo_color --switch=$(COLOR) --cyan "Run CPack packaging tool for source..."
	/usr/bin/cpack --config ./CPackSourceConfig.cmake /home/esofi/gzweb/build/CPackSourceConfig.cmake
.PHONY : package_source

# Special rule for the target package_source
package_source/fast: package_source

.PHONY : package_source/fast

# Special rule for the target rebuild_cache
rebuild_cache:
	@$(CMAKE_COMMAND) -E cmake_echo_color --switch=$(COLOR) --cyan "Running CMake to regenerate build system..."
	/usr/bin/cmake -S$(CMAKE_SOURCE_DIR) -B$(CMAKE_BINARY_DIR)
.PHONY : rebuild_cache

# Special rule for the target rebuild_cache
rebuild_cache/fast: rebuild_cache

.PHONY : rebuild_cache/fast

# Special rule for the target package
package: preinstall
	@$(CMAKE_COMMAND) -E cmake_echo_color --switch=$(COLOR) --cyan "Run CPack packaging tool..."
	/usr/bin/cpack --config ./CPackConfig.cmake
.PHONY : package

# Special rule for the target package
package/fast: package

.PHONY : package/fast

# The main all target
all: cmake_check_build_system
	$(CMAKE_COMMAND) -E cmake_progress_start /home/esofi/gzweb/build/CMakeFiles /home/esofi/gzweb/build/CMakeFiles/progress.marks
	$(MAKE) -f CMakeFiles/Makefile2 all
	$(CMAKE_COMMAND) -E cmake_progress_start /home/esofi/gzweb/build/CMakeFiles 0
.PHONY : all

# The main clean target
clean:
	$(MAKE) -f CMakeFiles/Makefile2 clean
.PHONY : clean

# The main clean target
clean/fast: clean

.PHONY : clean/fast

# Prepare targets for installation.
preinstall: all
	$(MAKE) -f CMakeFiles/Makefile2 preinstall
.PHONY : preinstall

# Prepare targets for installation.
preinstall/fast:
	$(MAKE) -f CMakeFiles/Makefile2 preinstall
.PHONY : preinstall/fast

# clear depends
depend:
	$(CMAKE_COMMAND) -S$(CMAKE_SOURCE_DIR) -B$(CMAKE_BINARY_DIR) --check-build-system CMakeFiles/Makefile.cmake 1
.PHONY : depend

#=============================================================================
# Target rules for targets named uninstall

# Build rule for target.
uninstall: cmake_check_build_system
	$(MAKE) -f CMakeFiles/Makefile2 uninstall
.PHONY : uninstall

# fast build rule for target.
uninstall/fast:
	$(MAKE) -f CMakeFiles/uninstall.dir/build.make CMakeFiles/uninstall.dir/build
.PHONY : uninstall/fast

#=============================================================================
# Target rules for targets named gzcoarse

# Build rule for target.
gzcoarse: cmake_check_build_system
	$(MAKE) -f CMakeFiles/Makefile2 gzcoarse
.PHONY : gzcoarse

# fast build rule for target.
gzcoarse/fast:
	$(MAKE) -f tools/CMakeFiles/gzcoarse.dir/build.make tools/CMakeFiles/gzcoarse.dir/build
.PHONY : gzcoarse/fast

# Help Target
help:
	@echo "The following are some of the valid targets for this Makefile:"
	@echo "... all (the default if no target is provided)"
	@echo "... clean"
	@echo "... depend"
	@echo "... edit_cache"
	@echo "... test"
	@echo "... package_source"
	@echo "... rebuild_cache"
	@echo "... package"
	@echo "... uninstall"
	@echo "... gzcoarse"
.PHONY : help

#=============================================================================
# Special targets to cleanup operation of make.

# Special rule to run CMake to check the build system integrity.
# No rule that depends on this can have commands that come from listfiles
# because they might be regenerated.
cmake_check_build_system:
	$(CMAKE_COMMAND) -S$(CMAKE_SOURCE_DIR) -B$(CMAKE_BINARY_DIR) --check-build-system CMakeFiles/Makefile.cmake 0
.PHONY : cmake_check_build_system
</details>

mis-eu, 2020-10-15:
> > which system releases are you on?
> 
> Gazebo 11 on Ubuntu Focal

@iche033 
@reinzor using your forked branch 
which **Node version** are you building with?

Node 6.14.4 doesn't support karma.conf.js
and fails to run:
get_local_models.py
webify_models.py
webify_models_v2.py

<h4>Build Log</h4>
<details>
usr@moon:~/gzweb$ npm run deploy --- -m local

> gzweb@1.3.0 deploy /home/usr/gzweb
> ./deploy.sh "-m" "local"

Build a local model database.
Only local models.
npm WARN deprecated istanbul@0.4.5: This module is no longer maintained, try this instead:
npm WARN deprecated   npm i nyc
npm WARN deprecated Visit https://istanbul.js.org/integrations for other alternatives.
npm WARN deprecated request@2.77.0: request has been deprecated, see https://github.com/request/request/issues/3142
npm WARN deprecated coffee-script@1.3.3: CoffeeScript on NPM has moved to "coffeescript" (no hyphen)
npm WARN deprecated minimatch@0.2.14: Please update to minimatch 3.0.2 or higher to avoid a RegExp DoS issue
npm WARN deprecated chokidar@1.7.0: Chokidar 2 will break on node v14+. Upgrade to chokidar 3 with 15x less dependencies.
npm WARN deprecated core-js@2.6.11: core-js@<3 is no longer maintained and not recommended for usage due to the number of issues. Please, upgrade your dependencies to the actual version of core-js@3.
npm WARN deprecated request@2.88.2: request has been deprecated, see https://github.com/request/request/issues/3142
npm WARN deprecated har-validator@2.0.6: this library is no longer supported
npm WARN deprecated hawk@3.1.3: This module moved to @hapi/hawk. Please make sure to switch over as this distribution is no longer supported and may contain bugs and critical security issues.
npm WARN deprecated node-uuid@1.4.8: Use uuid module instead
npm WARN deprecated graceful-fs@1.2.3: please upgrade to graceful-fs 4 for compatibility with current and future versions of Node.js
npm WARN deprecated fsevents@1.2.13: fsevents 1 will break on node v14+ and could be using insecure binaries. Upgrade to fsevents 2.
npm WARN deprecated har-validator@5.1.5: this library is no longer supported
npm WARN deprecated boom@2.10.1: This version has been deprecated in accordance with the hapi support policy (hapi.im/support). Please upgrade to the latest version to get the best features, bug fixes, and security patches. If you are unable to upgrade at this time, paid support is available for older versions (hapi.im/commercial).
npm WARN deprecated sntp@1.0.9: This module moved to @hapi/sntp. Please make sure to switch over as this distribution is no longer supported and may contain bugs and critical security issues.
npm WARN deprecated hoek@2.16.3: This version has been deprecated in accordance with the hapi support policy (hapi.im/support). Please upgrade to the latest version to get the best features, bug fixes, and security patches. If you are unable to upgrade at this time, paid support is available for older versions (hapi.im/commercial).
npm WARN deprecated cryptiles@2.0.5: This version has been deprecated in accordance with the hapi support policy (hapi.im/support). Please upgrade to the latest version to get the best features, bug fixes, and security patches. If you are unable to upgrade at this time, paid support is available for older versions (hapi.im/commercial).
npm WARN deprecated minimatch@0.3.0: Please update to minimatch 3.0.2 or higher to avoid a RegExp DoS issue
npm WARN deprecated circular-json@0.3.3: CircularJSON is in maintenance only, flatted is its successor.
npm WARN deprecated json3@3.3.2: Please use the native JSON object instead of JSON 3
npm WARN deprecated resolve-url@0.2.1: https://github.com/lydell/resolve-url#deprecated
npm WARN deprecated urix@0.1.0: Please see https://github.com/lydell/urix#deprecated

> bufferutil@4.0.1 install /home/usr/gzweb/node_modules/bufferutil
> node-gyp-build

> utf-8-validate@5.0.2 install /home/usr/gzweb/node_modules/utf-8-validate
> node-gyp-build

> core-js@2.6.11 postinstall /home/usr/gzweb/node_modules/core-js
> node -e "try{require('./postinstall')}catch(e){}"

Thank you for using core-js ( https://github.com/zloirock/core-js ) for polyfilling JavaScript standard library!

The project needs your help! Please consider supporting of core-js on Open Collective or Patreon: 
> https://opencollective.com/core-js 
> https://www.patreon.com/zloirock 

Also, the author of core-js ( https://github.com/zloirock ) is looking for a good job -)

npm notice created a lockfile as package-lock.json. You should commit this file.
npm WARN notsup Unsupported engine for karma@1.7.1: wanted: {"node":"0.10 || 0.12 || 4 || 5 || 6 || 7 || 8"} (current: {"node":"10.19.0","npm":"6.14.4"})
npm WARN notsup Not compatible with your version of node/npm: karma@1.7.1
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: fsevents@^1.0.0 (node_modules/chokidar/node_modules/fsevents):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for fsevents@1.2.13: wanted {"os":"darwin","arch":"any"} (current: {"os":"linux","arch":"x64"})
[... 1006 lines cut ...]
├───────────────┼──────────────────────────────────────────────────────────────┤
│ Patched in    │ > 4.2.0 < 5.0.0 || >= 5.0.3                                  │
├───────────────┼──────────────────────────────────────────────────────────────┤
│ Dependency of │ codecov [dev]                                                │
├───────────────┼──────────────────────────────────────────────────────────────┤
[... cut ...]
│ Path          │ codecov > request > hawk > cryptiles                         │
├───────────────┼──────────────────────────────────────────────────────────────┤
│ More info     │ https://npmjs.com/advisories/1464                            │
└───────────────┴──────────────────────────────────────────────────────────────┘
found 54 vulnerabilities (23 low, 11 moderate, 20 high) in 763 scanned packages
  48 vulnerabilities require semver-major dependency updates.
  6 vulnerabilities require manual review. See the full report for details.

mis-eu, 2020-10-15:
> Thanks for figuring this out. We could create a new branch that's compatible with Gazebo 11.
> 
> We should also update the tutorial: http://gazebosim.org/tutorials?tut=gzweb_install&cat=gzweb

@chapulina does gzweb require that user created robot models take the strict form of:
model.config   
model.sdf

or should gzweb be able to load them in this format:
model.config   #where <sdf version="1.5">**myrobot-customname**.sdf</sdf>
**myrobot-customname**.sdf

mis-eu, 2020-10-15:
> I applied @reinzor's fix in pull request #194. I didn't have to apply the `sed` boost cmd in `deploy.sh` and it still built so I left that out.

hi @iche033 
@chapulina 

please take a look below:

$ source /opt/ros/foxy/setup.bash
$ export GAZEBO_MODEL_PATH=$GAZEBO_MODEL_PATH:~/gzweb/http/client/assets
```npm run deploy --- -m local```
fails to run the deploy.sh script and the executable py scripts for generating the local robot model.

after updating the interpreter path in all the py scripts to:
#!/usr/bin/env python

and following up the incomplete
```npm run deploy --- -m local```
with a direct
``` bash deploy.sh```
the robot model does not render in the GUI

Is there a Node version incompatibility with Focal/Foxy/Gazebo11?
Would deploy.sh fail with certain size .STL or .DAE files?
Do the meshes have to be in .DAE format?

<h4>bash deploy.sh OUTPUT</4>  

<details>

esofi@moon:~/gzweb$ bash deploy.sh
npm WARN ajv-keywords@2.1.1 requires a peer of ajv@^5.0.0 but none is installed. You must install peer dependencies yourself.
npm WARN gzweb@1.3.0 No license field.
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: fsevents@1.2.13 (node_modules/fsevents):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for fsevents@1.2.13: wanted {"os":"darwin","arch":"any"} (current: {"os":"linux","arch":"x64"})

audited 763 packages in 7.895s

8 packages are looking for funding
  run `npm fund` for details

found 54 vulnerabilities (23 low, 11 moderate, 20 high)
  run `npm audit fix` to fix them, or `npm audit` for details
Running "concat:build_src" (concat) task

Running "concat:build_gz3d" (concat) task

Running "concat:build_gui" (concat) task

Running "jshint:files" (jshint) task
>> 2 files lint free.

Running "uglify:build_src" (uglify) task
>> 1 file created 1.48 MB → 693.93 kB

Running "uglify:build_gz3d" (uglify) task
>> 1 file created 1.48 MB → 693.93 kB

Running "uglify:build_gui" (uglify) task
>> 1 file created 2.13 MB → 1.14 MB

Done, without errors.
-- The C compiler identification is GNU 9.3.0
-- The CXX compiler identification is GNU 9.3.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
[... 250 lines cut ...]
gyp info spawn args   '/home/esofi/.cache/node-gyp/10.19.0/include/node/common.gypi',
gyp info spawn args   '-Dlibrary=shared_library',
gyp info spawn args   '-Dvisibility=default',
gyp info spawn args   '-Dnode_root_dir=/home/esofi/.cache/node-gyp/10.19.0',
gyp info spawn args   '-Dnode_gyp_dir=/home/esofi/gzweb/node_modules/node-gyp',
gyp info spawn args   '-Dnode_lib_file=/home/esofi/.cache/node-gyp/10.19.0/<(target_arch)/node.lib',
gyp info spawn args   '-Dmodule_root_dir=/home/esofi/gzweb/gzbridge',
gyp info spawn args   '-Dnode_engine=v8',
gyp info spawn args   '--depth=.',
gyp info spawn args   '--no-parallel',
gyp info spawn args   '--generator-output',
gyp info spawn args   'build',
gyp info spawn args   '-Goutput_dir=.' ]
gyp info ok 
gyp info it worked if it ends with ok
gyp info using node-gyp@7.1.1
gyp info using node@10.19.0 | linux | x64
gyp info spawn make
gyp info spawn args [ 'BUILDTYPE=Debug', '-C', 'build' ]
make: Entering directory '/home/esofi/gzweb/gzbridge/build'
make: Nothing to be done for 'all'.
make: Leaving directory '/home/esofi/gzweb/gzbridge/build'
gyp info ok 
Not cloning the model repo
Done

mis-eu, 2020-10-19:
@chapulina the Scene Properties can not be modified in the GUI
[image]

chapulina, 2020-11-11:
> should gzweb be able to load defined Scene Properties in the world.model from gzserver?

@mis-eu , mind ticketing a new issue for that? This issue is about compilation.

---

Closing this, compilation with Gazebo 11 is working on the `gzweb_1.4` branch since #194.

closed 2020-11-11 by chapulina