osrf/gzweb #185: Build error in docker container gazebo:libgazebo7

myFree, 2020-07-06:
Hi; apologies if I've missed relevant docs which explain how to avoid this.

I am trying to build the repo in a docker container
My Dockerfile is just:
```
FROM gazebo:libgazebo7
CMD [ "gzserver", "verbose" ]
```

For now; I'm manually installing the dependencies in the docker container

The build is erroring on

```
[100%] Built target gzcoarse
/root/gzweb/node_modules/env-paths/index.js:7
const {env} = process;
      ^

SyntaxError: Unexpected token {
    at exports.runInThisContext (vm.js:53:16)
    at Module._compile (module.js:374:25)
    at Object.Module._extensions..js (module.js:417:10)
    at Module.load (module.js:344:32)
    at Function.Module._load (module.js:301:12)
    at Module.require (module.js:354:17)
    at require (internal/module.js:12:17)
    at Object.<anonymous> (/root/gzweb/node_modules/node-gyp/bin/node-gyp.js:7:18)
    at Module._compile (module.js:410:26)
    at Object.Module._extensions..js (module.js:417:10)
/root/gzweb/node_modules/env-paths/index.js:7
const {env} = process;
      ^

SyntaxError: Unexpected token {
    at exports.runInThisContext (vm.js:53:16)
    at Module._compile (module.js:374:25)
    at Object.Module._extensions..js (module.js:417:10)
    at Module.load (module.js:344:32)
    at Function.Module._load (module.js:301:12)
    at Module.require (module.js:354:17)
    at require (internal/module.js:12:17)
    at Object.<anonymous> (/root/gzweb/node_modules/node-gyp/bin/node-gyp.js:7:18)
    at Module._compile (module.js:410:26)
    at Object.Module._extensions..js (module.js:417:10)
There are node-gyp build errors, exiting.

npm ERR! Linux 5.3.0-62-generic
npm ERR! argv "/usr/bin/nodejs" "/usr/bin/npm" "run" "deploy" "---" "-m"
npm ERR! node v4.2.6
npm ERR! npm  v3.5.2
npm ERR! code ELIFECYCLE
npm ERR! gzweb@1.3.0 deploy: `./deploy.sh "-m"`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the gzweb@1.3.0 deploy script './deploy.sh "-m"'.
npm ERR! Make sure you have the latest version of node.js and npm installed.
npm ERR! If you do, this is most likely a problem with the gzweb package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     ./deploy.sh "-m"
npm ERR! You can get information on how to open an issue for this project with:
npm ERR!     npm bugs gzweb
npm ERR! Or if that isn't available, you can get their info via:
npm ERR!     npm owner ls gzweb
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     /root/gzweb/npm-debug.log
```

myFree, 2020-07-07:
I must have been doing some silly mistake; working fine today; sorry for the false report

closed 2020-07-07 by myFree

vinnnyr, 2020-10-16:
Running into some similar issues, are you able to share or point me to your complete image or Dockerfile?