osrf/gzweb #193: Webifying breaks with model names with space in them

konduri, 2020-09-03:
Trying to webify a model call "floor_Living room"with following model.sdf breaks during webifying. The model comes up without any issue in gzclient.

```
<?xml version='1.0' encoding='us-ascii'?>
<sdf version="1.6">
  <model name="floor_Living room">
    <static>true</static>
    <link name="link">
      <collision name="collision">
        <pose>0 0 0 0 0 0</pose>
        <geometry>
          <mesh>
            <uri>model://generation_j592zpc9lvyz_world_1/floor_Living room/meshes/floor_Living room.dae</uri>
            <scale>1 1 1</scale>
          </mesh>
        </geometry>
      </collision>
      <visual name="visual">
        <meta>
          <layer>0</layer>
        </meta>
        <pose>0 0 0 0 0 0</pose>
        <geometry>
          <mesh>
            <uri>model://generation_j592zpc9lvyz_world_1/floor_Living room/meshes/floor_Living room.dae</uri>
            <scale>1 1 1</scale>
          </mesh>
        </geometry>
      </visual>
    </link>
  </model>
</sdf>
```

When webifying the above model, we encounter the following error. It appears that when parsing through the models, gzweb doesnt handle model names with spaces in them. "floor_living room" is considered as "room" and breaks the webification.

Specifically this line: https://github.com/osrf/gzweb/blob/master/webify_models_v2.py#L65
[image]

chapulina, 2020-09-03:
I think you may find many other issues trying to use spaces within names on Gazebo-classic. My recommendation is not to have spaces so you can avoid other problems. Here are some related issues:

* https://github.com/osrf/gazebo/issues/2056
* https://github.com/osrf/gazebo/issues/2369

(for reference, we have good support for names with spaces on Ignition)

chapulina, 2020-11-11:
This issue seems to be a duplicate of #160. I'll close this one and keep the conversation there.

closed 2020-11-11 by chapulina