osrf/gzweb #198: Gzweb 1.4.0 does not install globally - Dependency problems

DentOpt, 2020-12-27:
Hi all, 

I have tried to adapt gzweb and use it together with other JS applications. 
The idea is to create modified versions in a local repository of 
```
gz_bridge/server.js
http/client/index.html
```
For this purpose, I would like to create a new node project that uses the gzweb dependencies. 
I try to make these globally accessible via 
```
$ cd gzweb 
$ npm run deploy 
$ npm install -g 
```
and I get following output that indicates that "gzweb" :"1.4.0" should have been globally installed:
```
$ sudo npm install -g
npm WARN ajv-keywords@2.1.1 requires a peer of ajv@^5.0.0 but none is installed. You must install peer dependencies yourself.

+ gzweb@1.4.0
updated 1 package in 1.368s
```
However, if I switch to the new_project repository the global references are not found

```
$ npm install
npm ERR! 404  'gzweb@1.4.0' is not in the npm registry.
npm ERR! 404 You should bug the author to publish it (or use the name yourself!)
npm ERR! 404 It was specified as a dependency of 'new_project'
```

Can you recommend an global installation procedure for the gzweb dependencies?
Thanks in advance!

FelipeGdM, 2020-12-28:
Hi @DentOpt!

Are you going to use only gzweb dependencies or gzweb itself?

If you only need the dependencies, you may copy the dependencies list from package.json from gzweb to your project

If you need gzweb as a dependency of your project, you may install it from its Github repo, From [npm install docs](https://docs.npmjs.com/cli/v6/commands/npm-install)

```sh
npm install <git-host>:<git-user>/<repo-name>
```

In our case, it should be

```sh
npm install git://github.com:osrf/gzweb
```

Have you tried before these procedure of installing a package globally in order to reference it later in another project? I may be wrong, but I think it isn't what you want. From [npm docs](https://docs.npmjs.com/cli/v6/configuring-npm/folders)

>    Local install (default): puts stuff in ./node_modules of the current package root.
>    Global install (with -g): puts stuff in /usr/local or wherever node is installed.
>    Install it locally if you're going to require() it.
>    Install it globally if you're going to run it on the command line.
>    If you need both, then install it in both places, or use npm link.

Or, more specfic from [npm install docs](https://docs.npmjs.com/downloading-and-installing-packages-globally)

> Installing a package globally allows you to use the code in the package as a set of tools on your local computer.

Unless you are trying to use the python and bash scripts globally in your system, this isn't what you are looking for

DentOpt, 2021-01-04:
Hi Felipe, 
happy new year and thank you for your help!
That already brings me a bit further.
Following your advice I installed `npm install git://github.com:osrf/gzweb#gzweb_1.4` so the dependency is added to the package.json:

  "dependencies": {
     // other dependencies     ....
    "gzweb": "git://github.com:osrf/gzweb#gzweb_1.4"
  },

Going to the project directory, I use
`npm install`
The install is completing: 
```
npm install
npm WARN mise_gzweb@1.4.0 No license field.
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: fsevents@1.2.13 (node_modules/fsevents):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for fsevents@1.2.13: wanted {"os":"darwin","arch":"any"} (current: {"os":"linux","arch":"x64"})

audited 792 packages in 3.34s

9 packages are looking for funding
  run `npm fund` for details

found 54 vulnerabilities (23 low, 11 moderate, 20 high)
  run `npm audit fix` to fix them, or `npm audit` for details
```

However, if I load the module in a js file (executed with npm start), 
I still get for `const gzweb = require('gzweb');`: 
```
npm start
internal/modules/cjs/loader.js:638
    throw err;
    ^
Error: Cannot find module 'gzweb'
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! Exit status 1
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
```
Also  using `const gzweb = require('./node_modules/gzweb');` leads to the same issue

If I look at the folder structure it is there, also when I `npm list | grep gzweb` it is there:
```
project@0.0.1 /home/project_folder
├─┬ gzweb@1.4.0 (git://github.com/osrf/gzweb.git#47aa85407f94cdb049a2b84ce769243a6e09df5f)
```

Could it be that the target is not properly installed in the active branch 1.4.0?  
Do you have any suggestions how to debug this?

Thanks a lot for your help

FelipeGdM, 2021-01-05:
Happy new year!

I think I understoond what you want to do, but I'm afraid it won't be so simple. 

First things first, when you run `const gzweb = require('gzweb');` node will search for the `index.js` file in the root of the package, but there is no `index.js` in gzweb, so you receive an error of module not found ([what is require?](https://nodejs.org/en/knowledge/getting-started/what-is-require/) for more information)

In order to fix that, you may point to a specific file you want to import (i.e. gz3d/src/gzgui.js), but then we bump into another issue, there is no `module.exports` in any of the files in the project. `require()` will only return what is exported by the file, but nothing is exported. There is a reason to that, the project is structured to be build with Grunt.js and run in the client-side, not in the server-side with node.

So, that said, to do what you want to, you may add the `module.exports` in the files you want to import and even create an `index.js` in the root of the gzweb folder in to be able to import and use the classes/methods of the project

What classes/methods you want to use?

It may exist an easier solution, but I think there is no obvious way to do that. It may be the case of a small refactor to allow gzweb to be imported by nodejs applications and even publish it in npm after that

closed 2021-01-05 by DentOpt