osrf/gzweb #201: npm run deploy --- -m failed build

duwudi, 2021-01-05:
Followed [these instructions](http://gazebosim.org/gzweb.html#install-collapse-1) exactly and get the following error when running `npm run deploy --- -m`

My system setup: 
Ubuntu 20.04 LTS (inside multipass)
Gazebo 9 (as per instructions)
Node 6 (as per instructions)

What can I do to resolve this?

```
g++: fatal error: Killed signal terminated program cc1plus
compilation terminated.
make: *** [gzbridge.target.mk:126: Debug/obj.target/gzbridge/GazeboInterface.o] Error 1
make: Leaving directory '/home/ubuntu/gzweb/gzbridge/build'
gyp ERR! build error 
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/home/ubuntu/gzweb/node_modules/node-gyp/lib/build.js:194:23)
gyp ERR! stack     at emitTwo (events.js:106:13)
gyp ERR! stack     at ChildProcess.emit (events.js:191:7)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:219:12)
gyp ERR! System Linux 5.4.0-54-generic
gyp ERR! command "/home/ubuntu/.nvm/versions/node/v6.17.1/bin/node" "/home/ubuntu/gzweb/node_modules/.bin/node-gyp" "rebuild" "-d"
gyp ERR! cwd /home/ubuntu/gzweb/gzbridge
gyp ERR! node -v v6.17.1
gyp ERR! node-gyp -v v6.1.0
gyp ERR! not ok 
There are node-gyp build errors, exiting.

npm ERR! Linux 5.4.0-54-generic
npm ERR! argv "/home/ubuntu/.nvm/versions/node/v6.17.1/bin/node" "/home/ubuntu/.nvm/versions/node/v6.17.1/bin/npm" "run" "deploy" "---" "-m"
npm ERR! node v6.17.1
npm ERR! npm  v3.10.10
npm ERR! code ELIFECYCLE
npm ERR! gzweb@1.4.0 deploy: `./deploy.sh "-m"`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the gzweb@1.4.0 deploy script './deploy.sh "-m"'.
npm ERR! Make sure you have the latest version of node.js and npm installed.
npm ERR! If you do, this is most likely a problem with the gzweb package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     ./deploy.sh "-m"
npm ERR! You can get information on how to open an issue for this project with:
npm ERR!     npm bugs gzweb
npm ERR! Or if that isn't available, you can get their info via:
npm ERR!     npm owner ls gzweb
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     /home/ubuntu/gzweb/npm-debug.log

```

iche033, 2021-01-06:
hmm the output does not show the compilation error. Someone suggested this fix for focal: https://github.com/osrf/gzweb/issues/184#issuecomment-704398535. It's partially applied in the 1.4.1 branch. You can try also the `sed`  cmd in deploy.sh and see if that makes any difference.

kdoroszko, 2021-02-26:
Hello everyone,

I have the same problem. I looked for solution from branch 1.4 to 2.0.0. Last time I checked it on 1.4.0 and added this lines of code:
https://github.com/eurogroep/gzweb/commit/696148dc49b8a0ebd54d749c2e1a0d28189de120

I got this:
```
gyp ERR! build error 
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/home/administrator/gzweb/node_modules/node-gyp/lib/build.js:194:23)
gyp ERR! stack     at ChildProcess.emit (events.js:315:20)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:277:12)
gyp ERR! System Linux 5.8.0-44-generic
gyp ERR! command "/home/administrator/.nvm/versions/node/v14.16.0/bin/node" "/home/administrator/gzweb/node_modules/.bin/node-gyp" "build" "-d"
gyp ERR! cwd /home/administrator/gzweb/gzbridge
gyp ERR! node -v v14.16.0
gyp ERR! node-gyp -v v7.1.2
gyp ERR! not ok 
There are node-gyp build errors, exiting.
```

I also checked this:
https://github.com/Intelligent-Quads/iq_tutorials/blob/master/docs/gzweb_install.md

But there is no branch `fix_build_gz11`

My system:
Ubuntu 20.04
Gazebo11
Node 14.16.0

iche033, 2021-03-01:
the changes in `fix_build_gz11` branch have been merged in to `gzweb_1.4` branch. https://github.com/osrf/gzweb/pull/194

You can try the `gzweb_1.4` branch instead.

kdoroszko, 2021-03-04:
Like I wrote from 1.4 to 2.0.0 it doesn't work. Also on branch `gzweb_1.4`, still the same.

iche033, 2021-03-05:
can you try with node v12?