osrf/gzweb #202: DAE ERROR Load files in complex robots are renderd as the first dae loaded

RDaneelOlivav, 2021-01-07:
I'm executing thi in Ubuntu 20, noetic.
The dae files have this strange issue that the first dae loads perfectly, after that all the other just load as a glitched copy of the first.
Any ideas what could be happening here?

[image]
[image]

Tried to reexport the dae files but still happens.

RDaneelOlivav, 2021-01-11:
Fixed!

Its related to the materials and other names that are the same in all the daes. This normally is caused because you created and exported it piece by piece and all the elements use the same default material name ( for example in blender). In this newest version of gzweb its not dealt properly and it just inherits the first material into the rest of elements, glitching.

Solution: just edit the daes and change the material, UVmapping and shapename ) by unique names. That made the trick

[image]

duwudi, 2021-01-17:
@RDaneelOlivav I think I have the same problem, that's been causing me some confusion for a long time! Thanks a lot for the suggestion.

You mention the "newest version of gzweb", which gzweb version are you running? And are you aware of an older version where this isn't an issue?