osrf/gzweb #207: Uncaught DOMException: Failed to construct 'WebSocket': The URL 'ws://:' is invalid.     roslib.js:2224

newcanopies, 2021-04-05:
Hi

loading gzweb in electron (entrypoint: npm start via gzbridge/server.js) the websocket server is not streaming headless gzserver
failing to construct the websocket at roslib.js:2224

`Ros.prototype.__proto__ = EventEmitter2.prototype;

/**
 * Connect to the specified WebSocket.
 *
 * @param url - WebSocket URL or RTCDataChannel label for Rosbridge
 */
Ros.prototype.connect = function(url) {
  if (this.transportLibrary === 'socket.io') {
    this.socket = assign(io(url, {'force new connection': true}), socketAdapter(this));
    this.socket.on('connect', this.socket.onopen);
    this.socket.on('data', this.socket.onmessage);
    this.socket.on('close', this.socket.onclose);
    this.socket.on('error', this.socket.onerror);
  } else if (this.transportLibrary.constructor.name === 'RTCPeerConnection') {
    this.socket = assign(this.transportLibrary.createDataChannel(url, this.transportOptions), socketAdapter(this));
  } else {
    if (!this.socket || this.socket.readyState === WebSocket.CLOSED) {
      var sock = new WebSocket(url);
      sock.binaryType = 'arraybuffer';
      this.socket = assign(sock, socketAdapter(this));
    }
  }

};`

how to bridge gzweb websockets within electron app? 
Thank you!

newcanopies, 2021-04-05:
npm start --display drm --device /dev/dri/renderiHD  --enable-debug=3

```

> gzweb@1.4.0 start /home/user/gzweb
> if [ $npm_config_port ]; then port=$npm_config_port; fi; cd gzbridge && ./server.js $port "drm" "/dev/dri/renderiHD"

Mon Apr 05 2021 15:51:59 GMT+0200 (Central European Summer Time) Static server listening on port: drm
boost::filesystem::directory_iterator::construct: No such file or directory: "/assets"
--------------------------------------------------------------
Gazebo transport node connected to gzserver.
Pose message filter parameters between successive messages: 
  minimum seconds: 0.02
  minimum XYZ distance squared: 0.00001
  minimum Quartenion distance squared: 0.00001
--------------------------------------------------------------
events.js:291
      throw er; // Unhandled 'error' event
      ^

at Server.setupListenHandle [as _listen2] (net.js:1299:21)
    at listenInCluster (net.js:1364:12)
    at Server.listen (net.js:1461:5)
    at Object.<anonymous> (/home/user/gzweb/gzbridge/server.js:345:12)
    at Module._compile (internal/modules/cjs/loader.js:999:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:1027:10)
    at Module.load (internal/modules/cjs/loader.js:863:32)
    at Function.Module._load (internal/modules/cjs/loader.js:708:14)
    at Function.executeUserEntryPoint [as runMain] (internal/modules/run_main.js:60:12)
    at internal/main/run_main_module.js:17:47
Emitted 'error' event on Server instance at:
    at emitErrorNT (net.js:1343:8)
    at processTicksAndRejections (internal/process/task_queues.js:84:21) {
  code: 'EADDRINUSE',
  errno: 'EADDRINUSE',
  syscall: 'listen',
  address: 'drm',
  port: -1
}
Error: listen EADDRINUSE: address already in use drm
node: /usr/include/boost/smart_ptr/shared_ptr.hpp:734: typename boost::detail::sp_member_access<T>::type boost::shared_ptr<T>::operator->() const [with T = gazebo::transport::Publication; typename boost::detail::sp_member_access<T>::type = gazebo::transport::Publication*]: Assertion `px != 0' failed.
node: /usr/include/boost/smart_ptr/shared_ptr.hpp:734: typename boost::detail::sp_member_access<T>::type boost::shared_ptr<T>::operator->() const [with T = gazebo::transport::Publication; typename boost::detail::sp_member_access<T>::type = gazebo::transport::Publication*]: Assertion `px != 0' failed.
node: /usr/include/boost/smart_ptr/shared_ptr.hpp:734: typename boost::detail::sp_member_access<T>::type boost::shared_ptr<T>::operator->() const [with T = gazebo::transport::Publication; typename boost::detail::sp_member_access<T>::type = gazebo::transport::Publication*]: Assertion `px != 0' failed.
```

newcanopies, 2021-04-05:
both in electron and in browser at localhost:8080
only the static HTML content is served 
(with the exception of the GRID lines and local model)

[image]

iche033, 2021-04-06:
I have not used electron before so not sure what the actual issue is but from looking at the error msg in the title, it seems like this line is the issue:
https://github.com/osrf/gzweb/blob/master/gz3d/src/gziface.js#L13

the `location` object seems to be empty?

newcanopies, 2021-04-06:
>  it seems like this line is the issue:
> https://github.com/osrf/gzweb/blob/master/gz3d/src/gziface.js#L13

thank you @iche033 

since gzweb runs on localhost:8080
hostname = localhost
port = 8080

```
this.url = url ||
      (location.hostname + ':' + location.port + location.pathname);
```

```
GZ3D.GZIface.prototype.connect = function()
{
  this.webSocket = new ROSLIB.Ros({
    url : (this.secure ? 'wss://' : 'ws://') + this.url
  });
```

**where is the location object defined in the code?**

**when/how during npm start is gziface.js called along with other scripts under gzweb/gz3d/src?**

newcanopies, 2021-04-06:
Shouldn't this run through the script by itself in order to debug?

$ node gziface.js 

```
/home/user/gzweb/gz3d/src/gziface.js:15
GZ3D.GZIface = function(scene, url)
^

ReferenceError: GZ3D is not defined
    at Object.<anonymous> (/home/user/gzweb/gz3d/src/gziface.js:15:1)
    at Module._compile (internal/modules/cjs/loader.js:999:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:1027:10)
    at Module.load (internal/modules/cjs/loader.js:863:32)
    at Function.Module._load (internal/modules/cjs/loader.js:708:14)
    at Function.executeUserEntryPoint [as runMain] (internal/modules/run_main.js:60:12)
    at internal/main/run_main_module.js:17:47
```

iche033, 2021-04-06:
the `location`object should be supported on most browsers, https://www.w3schools.com/jsref/prop_loc_hostname.asp

here's where all the gz* objects are created: https://github.com/osrf/gzweb/blob/master/gz3d/client/index.html#L1101-L1105

Those are client side javascript files. The entry point is index.html

newcanopies, 2021-04-10:
hi @iche033 

I was able to wrap gzweb in a react-app, at least to the extent of fully loading THREE.WebGLRenderer (ie, world gridlines load as well).

However, even after implementing openssl HTTPS/wss connection, and propagating it throughout the package, the same
WebSocket connection to 'wss://localhost:8080/' failed:  Error in connection establishment: net::ERR_SSL_PROTOCOL_ERROR

is thrown (in chrome browser) at roslib.js:2224
which can be seen here at [roslib.js:2307](https://github.com/RobotWebTools/roslibjs/blob/2e125d02534566fede8bec03feb3a29ba5b38a72/build/roslib.js#L2307)

does roslibjs `WebSocket` support` wss?`   
is the `WebSocket` module used by roslibjs a `Node` or `third-party` websocket module?

throughout the gzweb package are WebSocket and WebSocket Server consistently require the same module?

thank you!

newcanopies, 2021-04-11:
https://github.com/RobotWebTools/roslibjs/issues/53#issue-23876804

iche033, 2021-04-12:
we're using an old version of roslib.js:
https://github.com/osrf/gzweb/blob/master/gz3d/client/js/include/roslib.js

We are using it on the client side only and it's just using the WebSocket API from the browser:
https://github.com/osrf/gzweb/blob/master/gz3d/client/js/include/roslib.js#L431

I have not tried with `wss` but I think you'll also need to modify gzbridge's server.js to open a secure connection instead of an unsafe one:
https://github.com/osrf/gzweb/blob/master/gzbridge/server.js 

It's been a while but I think this is the one we're using on the server side:
https://github.com/theturtle32/WebSocket-Node

closed 2021-04-20 by newcanopies