osrf/gzweb #208: node: /usr/include/boost/smart_ptr/shared_ptr.hpp:734: typename boost::detail::sp_member_access<T>::type

newcanopies, 2021-04-23:
Hi everyone, 

I have two ROS packages, each running a websocket server on the same PORT.

Is that the conflict throwing this boost:: error?  
`
node: /usr/include/boost/smart_ptr/shared_ptr.hpp:734: typename boost::detail::sp_member_access<T>::type boost::shared_ptr<T>::operator->() const [with T = gazebo::transport::Publication; typename boost::detail::sp_member_access<T>::type = gazebo::transport::Publication*]: Assertion `px != 0' failed.`

What does **Publication; typename boost::detail::sp_member_access<T>::type = gazebo::transport::Publication*]: Assertion `px != 0' failed.`** indicate?

Is there a websocket server configuration that can allow two different ROS nodes to listen on the same PORT?

Thank you

newcanopies, 2021-04-24:
is it a threading issue?

newcanopies, 2021-04-24:
is it related to https://answers.gazebosim.org/question/7173/gazebo-crashes-assertion-px-0-failed/?answer=25951#post-id-25951
 >errors usually come from trying to access joints or links through pointer that do not exist.

newcanopies, 2021-04-24:
this is the boost line throwing the **Assertion px** error
[smart_ptr/include/boost/smart_ptr/shared_ptr.hpp](https://github.com/boostorg/smart_ptr/blob/f3424e74e83821ebd616dc3e54c9c1f426af3886/include/boost/smart_ptr/shared_ptr.hpp#L732-L735)

```
    typename boost::detail::sp_array_access< T >::type operator[] ( std::ptrdiff_t i ) const BOOST_SP_NOEXCEPT_WITH_ASSERT
    {
        BOOST_ASSERT( px != 0 );
        BOOST_ASSERT( i >= 0 && ( i < boost::detail::sp_extent< T >::value || boost::detail::sp_extent< T >::value == 0 ) );
```

Jinsun-Lee, 2023-11-20:
`source /usr/share/gazebo/setup.bash`
`source /usr/share/gazebo-11/setup.bash`

This worked for me.

https://robotics.stackexchange.com/questions/104352/how-to-analyse-this-gazebo-with-turtlebot3-on-ros2-rolling