osrf/gzweb #209: URDF support in GzWeb?

newcanopies, 2021-05-03:
Hi there, 

I am controlling a URDF robot in Gazebo with ` libgazebo_ros_control.so`. 

How can GzWeb load this URDF?
If GzWeb only supports SDF, is there a workaround to load the URDF running on gzserver also in GzWeb?

FelipeGdM, 2021-05-05:
Hi!

There are some workarounds that you may try. The easiest one is to manually convert your `.urdf` model to `.sdf` (Gazebo does it automatically under the hook as said in the [ROS urdf tutorial](http://gazebosim.org/tutorials/?tut=ros_urdf)). Then you can create a `model.config` file and save your model in your local model library (some folder listed in `GAZEBO_MODEL_PATH`)

You can do that with the command `gz sdf`

```bash
gz sdf -p my_robot.urdf > my_robot.sdf
```

The downside of this approach is maintaining duplicated information, you have the same robot model in `.urdf` and `.sdf` stored in your project 

The other option is given in the [roslaunch Gazebo tutorial](gazebosim.org/tutorials?tut=ros_roslaunch) (see "Model Database" Robot Spawn Method) and consists in treating the robot description repository as a model database. Quoting the docs

>  This isn't the cleanest way to load your URDF into Gazebo but accomplishes the goal of not having to keep two copies of your robot URDF on your computer.

Both approaches require the creation of a Gazebo model of your robot, so it is recognized as an asset to be loaded by gzweb

newcanopies, 2021-05-05:
Thanks very much for your thoughtful response FelipeGdM!

The problem is that I went through all that documentatio already, but there are a few deadlock issues:

* re. converting to Gazebo native SDF:  
can SDF <include> Joint Controllers from `moveit2` and interpret KDL tree kinematics? 
For kinematic control purposes URDF may be necessary.

* re. spawning:  
`spawn_entity.py` is not able to publish the model meshes. Here the UR10 model links and joints (published by robot_state_publisher from` robot_description`) appear in Gazebo, but not the visual and collision meshes https://github.com/ros-simulation/gazebo_ros_pkgs/issues/1272 

Is the bottom line that gzweb cannot use URDF, 
only SDF?

FelipeGdM, 2021-05-07:
I'm affraid I didn't quite understand your problem. I mean, we are dealing with 2 separated entities, gzserver ("backend" - runs the simulation itsef) and gzweb ("frontend" - renders the simulation). The problem is happening when you start your environment with gzserver or when you try to visualize it with gzweb?

I faced a similar rendering issue when I tried to visualize an environment with a URDF robot. Both gzserver and gzclient worked fine, but gzweb couldn't render the robots `.stl` meshes. The problem was that the robot was not listed as Gazebo model (because it was a ROS URDF model) and so its meshes were not loaded (i.e. copied to `assets/`) during gzweb's deploy process (the problem could be seen with 404 errors in browser console). Both solutions I mentioned worked in my case and eventually I decided to keep a "visualization model" of my robot in `.sdf` inside `models/` folder (alongside its meshes) and a "functional model" of my robot in `.urdf` inside its `robot_description` repo. 

As my environment worked with gzserver and gzclient, I knew it was a gzweb-related issue. In your case, it seems that the problem is within the backend part of the process, not gzweb itself

closed 2021-05-07 by newcanopies