osrf/gzweb #216: `npm run deploy` fails on gazebo11 and node 16.2

ryanhugh, 2021-06-03:
I'm on:
MacOS 11.4,
Gazebo11 (same error on gazebo9)
Node.js 16.2.0

Outside of this error, does this package support MacOS or gazebo11? 

When running: `npm run deploy --- -m` 

```
In file included from ../GZNode.cc:21:
In file included from ../GazeboInterface.hh:30:
In file included from /usr/local/lib/pkgconfig/../..//include/gazebo-11/gazebo/transport/TransportIface.hh:27:
In file included from /usr/local/lib/pkgconfig/../..//include/gazebo-11/gazebo/transport/Node.hh:30:
In file included from /usr/local/lib/pkgconfig/../..//include/gazebo-11/gazebo/transport/TopicManager.hh:35:
In file included from /usr/local/lib/pkgconfig/../..//include/gazebo-11/gazebo/transport/SubscriptionTransport.hh:24:
/usr/local/lib/pkgconfig/../..//include/gazebo-11/gazebo/transport/Connection.hh:57:65: error: expected class name
    class GZ_TRANSPORT_VISIBLE ConnectionReadTask : public tbb::task
                                                                ^
/usr/local/lib/pkgconfig/../..//include/gazebo-11/gazebo/transport/Connection.hh:73:20: error: no type named 'task' in namespace 'tbb'
      public: tbb::task *execute()
              ~~~~~^
/usr/local/lib/pkgconfig/../..//include/gazebo-11/gazebo/transport/Connection.hh:313:61: error: no member named 'allocate_root' in namespace 'tbb::v1::task'
                  ConnectionReadTask *task = new(tbb::task::allocate_root())
                                                 ~~~~~~~~~~~^
/usr/local/lib/pkgconfig/../..//include/gazebo-11/gazebo/transport/Connection.hh:315:19: error: no type named 'enqueue' in namespace 'tbb::v1::task'; did you mean 'ifqueue'?
                  tbb::task::enqueue(*task);
                  ^~~~~~~~~~~~~~~~~~
                  ifqueue
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/net/if_var.h:230:9: note: 'ifqueue' declared here
struct  ifqueue {
        ^
In file included from ../GZNode.cc:21:
In file included from ../GazeboInterface.hh:30:
In file included from /usr/local/lib/pkgconfig/../..//include/gazebo-11/gazebo/transport/TransportIface.hh:27:
/usr/local/lib/pkgconfig/../..//include/gazebo-11/gazebo/transport/Node.hh:39:58: error: expected class name
    class GZ_TRANSPORT_VISIBLE PublishTask : public tbb::task
                                                         ^
/usr/local/lib/pkgconfig/../..//include/gazebo-11/gazebo/transport/Node.hh:54:20: error: no type named 'task' in namespace 'tbb'
      public: tbb::task *execute()
              ~~~~~^
/usr/local/lib/pkgconfig/../..//include/gazebo-11/gazebo/transport/Node.hh:162:52: error: no member named 'allocate_root' in namespace 'tbb::v1::task'
                PublishTask *task = new(tbb::task::allocate_root())
                                        ~~~~~~~~~~~^
/usr/local/lib/pkgconfig/../..//include/gazebo-11/gazebo/transport/Node.hh:165:17: error: no type named 'enqueue' in namespace 'tbb::v1::task'; did you mean 'ifqueue'?
                tbb::task::enqueue(*task);
                ^~~~~~~~~~~~~~~~~~
                ifqueue
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/net/if_var.h:230:9: note: 'ifqueue' declared here
struct  ifqueue {
        ^
28 warnings and 8 errors generated.
make: *** [Debug/obj.target/gzbridge/GZNode.o] Error 1
gyp ERR! build error
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/Users/ryan/Desktop/code/subway_simulation/gzweb/node_modules/node-gyp/lib/build.js:194:23)
gyp ERR! stack     at ChildProcess.emit (node:events:365:28)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (node:internal/child_process:290:12)
gyp ERR! System Darwin 20.5.0
gyp ERR! command "/usr/local/Cellar/node/16.2.0/bin/node" "/Users/ryan/Desktop/code/subway_simulation/gzweb/node_modules/.bin/node-gyp" "rebuild" "-d"
gyp ERR! cwd /Users/ryan/Desktop/code/subway_simulation/gzweb/gzbridge
gyp ERR! node -v v16.2.0
gyp ERR! node-gyp -v v6.1.0
gyp ERR! not ok
There are node-gyp build errors, exiting. 
```

iche033, 2021-06-03:
I think the gazebo bottle is fixed but it could still be affecting downstream libraries that depend on gazebo. Possible workaround: https://github.com/PX4/PX4-Autopilot/issues/17644#issuecomment-847638715

ryanhugh, 2021-06-03:
Hmm. Diving into it a bit more it looks like running: 

``` 
brew update
brew install gazebo11
``` 

will install both tbb (the incorrect one) and tbb@2020_u3 (the correct one). It looks like `tbb` becomes the default, and files like this: `/usr/local/include/oneapi/tbb/version.h` (included by `/usr/local/include/tbb/version.h`) are from `tbb` instead of `tbb@2022_u3`. Looks like this "default" file is picked up when running `npm run deploy --- -m`, which causes this error. The fix from https://github.com/PX4/PX4-Autopilot/issues/17644#issuecomment-847638715 works as a one-off temporary work around. 

Is there is a way to point this package to use the header files in `/usr/local/Cellar/tbb@2020_u3/2020_U3/include/tbb/` instead of `/usr/local/include/tbb` until gazebo/this package is updated to use the latest version of tbb? :) 

I got another error after that one too: 

``` 
29 warnings generated.
  CXX(target) Debug/obj.target/gzbridge/pb2json.o
  CXX(target) Debug/obj.target/gzbridge/ConfigLoader.o
  CXX(target) Debug/obj.target/gzbridge/OgreMaterialParser.o
  SOLINK_MODULE(target) Debug/gzbridge.node
ld: library not found for -lboost_thread
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make: *** [Debug/gzbridge.node] Error 1
gyp ERR! build error
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/myproject/gzweb/node_modules/node-gyp/lib/build.js:194:23)
gyp ERR! stack     at ChildProcess.emit (node:events:394:28)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (node:internal/child_process:290:12)
gyp ERR! System Darwin 20.5.0
gyp ERR! command "/usr/local/Cellar/node/16.3.0/bin/node" "/myproject/gzweb/node_modules/.bin/node-gyp" "rebuild" "-d"
gyp ERR! cwd /myproject/gzweb/gzbridge
gyp ERR! node -v v16.3.0
gyp ERR! node-gyp -v v6.1.0
gyp ERR! not ok
There are node-gyp build errors, exiting.
sh: -m: command not found
``` 

Looks like the same issue from here: https://github.com/facebook/redex/issues/292

``` 
~/D/c/s/gzweb (master)> ls  /usr/local/lib/libboost_thread*
/usr/local/lib/libboost_thread-mt.a     /usr/local/lib/libboost_thread-mt.dylib
```  

Looks like updating gzbridge.target.mk temporarily fixes this issue. 

So to fix this second issue & to get it working on mac you have to : 

1) `npm run deploy --- -m` 
2) Update gzbridge/build/gzbridge.target.mk and change `-lboost_thread` to '-lboost_thread-mt`
3) cd gzbridge/build
4) run make
5) Comment out lines 65 to 95 in deploy.sh.
6) run deploy.sh again. 
7) `npm run start`