osrf/gzweb #226: gzweb models white or non-existent (both custom and standard)

IanBurwell, 2021-11-02:
I want to use gzweb with a custom docker image I've been using for a while, as gzclient and x-passthrough are quite slow on windows (WSL technically, but still). I have followed the tutorial linked in the README and have set `GAZEBO_MODEL_PATH` to include my custom models. When I connect to gzweb, however, my world is completely textureless, my robot is invisible, and other models I add in seem to have a 50% chance of having textures. Where did I go wrong?

Pictures:
[image]
[image]
[image]

Only error messages I could find were of the form:
[image]

My system: 
- Docker (on Windows 11) running Ubuntu 20
- Chrome (On Windows side)
- Installed branch `gzweb_1.4.1` and gazebo 11.7.0

iche033, 2021-11-02:
looks like the missing textures are the ones distributed by gazebo. It could be that the `media` directory is not copied into gzweb's assets folder. One thing to check is that after sourcing the `setup.sh` file, make sure that gazebo's `media` directory is in `GAZEBO_RESOURCE_PATH` before running the deploy step.

One manual workaround is just to manually copy the `media` folder as suggested in:
https://osrf-migration.github.io/gzweb-gh-pages/#!/osrf/gzweb/issues/61/page/1

IanBurwell, 2021-11-04:
On further inspection, these were not copied over due to gweb not finding python at `/usr/bin/python`. I created a symlink between there and `/usr/bin/python2.7` and it seems to run the scripts now. When building, I run into the below error, but most functionality seems to work. Now, however, I am getting black objects. Any ideas? 

```
convert-im6.q16: unable to open image `Color.png': No such file or directory @ error/blob.c/OpenBlob/2874.
convert-im6.q16: no images defined `/Color.png' @ error/convert.c/ConvertImageCommand/3258.
error Command '['convert', 'Color.png', '/Color.png']' returned non-zero exit status 1
Traceback (most recent call last):
  File "./webify_models_v2.py", line 44, in <module>
    subprocess.check_call(cmd)
  File "/usr/lib/python2.7/subprocess.py", line 190, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['convert', 'Color.png', '/Color.png']' returned non-zero exit status 1
Done
```

[image]

iche033, 2021-11-05:
looks like some textures failed to convert / copy over to the assets directory. If you open the browser developer tools and look the console tab, it may show that resources failed to load and where gzweb expects to download them from. You can manually copy those textures over to the corresponding model folder in `httpt/client/assets` and restart gzweb. You can ignore the `0.png` ones if they exist in the console as they are related to thumbnails.

IanBurwell, 2023-07-26:
I am no longer working on the project where I had this issue. Closing for now.

closed 2023-07-26 by IanBurwell