osrf/gzweb #240: Why is gazebo running smoothly and gzweb too laggy. Is there a problem with my use? Can someone answer it? thank you

HEFUHUI, 2022-09-20:
Why is gazebo running smoothly and gzweb too laggy. Is there a problem with my use? Can someone answer it? thank you

codebot, 2022-09-20:
Can you describe your use case in more detail? For example, what kind of robot(s), what kind of environment, what size of model(s) in terms of links and mesh complexity, etc.

HEFUHUI, 2022-09-27:
挺复杂的，是一辆小车来的，大概有77个组件构成。有行走、抓去物品等功能，目前在 gazebo_client 中很流畅，而在gzweb中很卡

HEFUHUI, 2022-09-27:
It's very complicated. It's a small car. It's made up of 77 components. It has the function of walking and grabbing objects. It is currently in gazebo_ It is very smooth in client, but too laggy in gzweb

codebot, 2022-09-27:
In general, WebGL and web browsers generally will have reduced performance rendering complex models, as opposed to "normal" desktop applications, because browser-based applications have to run in a sandbox with many restrictions. I would suggest trying to reduce the polygon count as much as possible, as an experiment, to see if that helps. Skilled 3D artists can do amazing tricks so that complex-looking models can be represented by very few polygons.

How many triangles are currently in your meshes?

HEFUHUI, 2022-09-28:
More than 80

closed 2024-03-31 by HEFUHUI