osrf/gzweb #249: How to automatically change the angle of view of Gazebo in Gzweb?

anducnguyen, 2023-07-12:
Hi guys,

I am porting my simulation to Gzweb. On the local machine, I can modify the GUI by code in the **.world** file to adjust the starting angle to the desired angle whenever Gazebo is launched. However, when I port the same simulation to Gzweb, the angle of view of my world model remains fixed at the default angle. Is there a way to modify the angle of view of the simulation in Gzweb automatically, without having to use the scroll click button on the mouse to change the angle?
Thanks.

from this:
[image]

to that:
[image]

iche033, 2023-07-12:
you maybe able to just modify the default camera position: https://github.com/osrf/gzweb/blob/1082133c7158c1dab30b29b59a3f6cf37cce3e21/gz3d/src/gzscene.js#L57

anducnguyen, 2023-07-14:
Thanks a lot for the suggestion, IC! The problem is that even when I tried to change:
>THREE.Vector3(0, -n, n);
and re-port it to gzweb by running `npm start`,
the angle stays the same as the default. Nothing has changed. It seems that normally, I would need to change a vector of 6 (x, y, z, yaw, roll, pitch) of the camera, but I could not find it anywhere!
Would you recommend any other ways?

closed 2023-07-14 by anducnguyen